apache/mynewt-core #2583: Error compiling 1.9.0

danielkucera, 2021-04-26:
`tinycbor` fails to compile in 1.9.0
```
+ newt build gas_meter
Building target targets/gas_meter
Compiling apps/gas_meter/src/eeprom.c
Compiling apps/gas_meter/src/main.c
Compiling apps/gas_meter/src/utils.c
Compiling repos/apache-mynewt-core/encoding/base64/src/hex.c
Compiling repos/apache-mynewt-core/encoding/base64/src/base64.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cbor_buf_reader.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cbor_buf_writer.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cbor_mbuf_reader.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cbor_mbuf_writer.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cborencoder.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cborencoder_close_container_checked.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cborerrorstrings.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cborparser.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cborparser_dup_string.c
Compiling repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c
Error: repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c: In function 'value_to_pretty':
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:307:33: error: expected ')' before 'PRIu64'
  307 |             if (fprintf(out, "%" PRIu64, val) < 0)
      |                                 ^~~~~~~
      |                                 )
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:307:31: error: spurious trailing '%' in format [-Werror=format=]
  307 |             if (fprintf(out, "%" PRIu64, val) < 0)
      |                               ^
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:313:38: error: expected ')' before 'PRIu64'
  313 |                 if (fprintf(out, "-%" PRIu64, val) < 0)
      |                                      ^~~~~~~
      |                                      )
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:313:36: error: spurious trailing '%' in format [-Werror=format=]
  313 |                 if (fprintf(out, "-%" PRIu64, val) < 0)
      |                                    ^
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:357:29: error: expected ')' before 'PRIu64'
  357 |         if (fprintf(out, "%" PRIu64 "(", tag) < 0)
      |                             ^~~~~~~
      |                             )
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:357:27: error: spurious trailing '%' in format [-Werror=format=]
  357 |         if (fprintf(out, "%" PRIu64 "(", tag) < 0)
      |                           ^
cc1: all warnings being treated as errors
```
If I switch back to 1.8.0, everything works again.

sjanc, 2021-04-27:
this is probably due to c1f0300413fc981741d37d655654de1dbca5d4df

What SDK do you use? I recall gcc-arm-none-eabi available in ubuntu had packaging issues resulting in this error. Could you try with ARM's gcc?

danielkucera, 2021-04-27:
I'm not much in favor of keeping a separate toolchain because of mynewt, is there not any other way around?
```
danman@silverhorse:~$ which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc
danman@silverhorse:~$ /usr/bin/arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/9.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --enable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:9-2019-q4-0ubuntu1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1) 
```

sjanc, 2021-04-27:
Do you mean you are able to compile other projects that use PRIuX macros with that SDK?

Those macros should be defined in inttypes.h provided by SDK, if those are missing you may define those yourself ( #define PRIu64 "llu"    should do)

danielkucera, 2021-04-27:
These macros were already present in 1.8.0:
https://github.com/apache/mynewt-core/blob/mynewt_1_8_0_tag/encoding/tinycbor/src/cborpretty.c#L307

And this one compiles without issues. So I presume the problem won't be in my SDK, or?

sjanc, 2021-04-27:
as mentioned (c1f0300), we now take this from SDK instead of local copy (which was wrong for some platforms)

danielkucera, 2021-04-27:
Okay, now I'm starting to understand. This has worked for me as a workaround:
```
sudo apt install libnewlib-nano-arm-none-eabi
newt target amend nrf52_blinky cflags="-I/usr/lib/newlib-nano/arm-none-eabi/include"
```

Now I'm hitting another error:
```
Compiling repos/apache-mynewt-core/libc/baselibc/src/memccpy.c
Error: repos/apache-mynewt-core/libc/baselibc/src/malloc.c:30:8: error: unknown type name 'malloc_lock_t'
   30 | static malloc_lock_t malloc_lock = &malloc_lock_nop;
      |        ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:30:36: error: initialization of 'int' from '_Bool (*)()' makes integer from pointer without a cast [-Werror=int-conversion]
   30 | static malloc_lock_t malloc_lock = &malloc_lock_nop;
      |                                    ^
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:31:8: error: unknown type name 'malloc_unlock_t'
   31 | static malloc_unlock_t malloc_unlock = &malloc_unlock_nop;
      |        ^~~~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:31:40: error: initialization of 'int' from 'void (*)()' makes integer from pointer without a cast [-Werror=int-conversion]
   31 | static malloc_unlock_t malloc_unlock = &malloc_unlock_nop;
      |                                        ^
repos/apache-mynewt-core/libc/baselibc/src/malloc.c: In function 'malloc':
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:159:10: error: called object 'malloc_lock' is not a function or function pointer
  159 |     if (!malloc_lock())
      |          ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:30:22: note: declared here
   30 | static malloc_lock_t malloc_lock = &malloc_lock_nop;
      |                      ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:175:13: error: implicit declaration of function 'add_malloc_block' [-Werror=implicit-function-declaration]
  175 |             add_malloc_block(more_mem, size);
      |             ^~~~~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:179:5: error: called object 'malloc_unlock' is not a function or function pointer
  179 |     malloc_unlock();
      |     ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:31:24: note: declared here
   31 | static malloc_unlock_t malloc_unlock = &malloc_unlock_nop;
      |                        ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c: At top level:
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:184:6: error: conflicting types for 'add_malloc_block' [-Werror]
  184 | void add_malloc_block(void *buf, size_t size)
      |      ^~~~~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:175:13: note: previous implicit declaration of 'add_malloc_block' was here
  175 |             add_malloc_block(more_mem, size);
      |             ^~~~~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c: In function 'add_malloc_block':
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:198:10: error: called object 'malloc_lock' is not a function or function pointer
  198 |     if (!malloc_lock())
      |          ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:30:22: note: declared here
   30 | static malloc_lock_t malloc_lock = &malloc_lock_nop;
      |                      ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:221:5: error: called object 'malloc_unlock' is not a function or function pointer
  221 |     malloc_unlock();
      |     ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:31:24: note: declared here
   31 | static malloc_unlock_t malloc_unlock = &malloc_unlock_nop;
      |                        ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c: In function 'free':
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:238:10: error: called object 'malloc_lock' is not a function or function pointer
  238 |     if (!malloc_lock())
      |          ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:30:22: note: declared here
   30 | static malloc_lock_t malloc_lock = &malloc_lock_nop;
      |                      ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:243:5: error: called object 'malloc_unlock' is not a function or function pointer
  243 |     malloc_unlock();
      |     ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:31:24: note: declared here
   31 | static malloc_unlock_t malloc_unlock = &malloc_unlock_nop;
      |                        ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c: In function 'get_malloc_memory_status':
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:252:10: error: called object 'malloc_lock' is not a function or function pointer
  252 |     if (!malloc_lock())
      |          ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:30:22: note: declared here
   30 | static malloc_lock_t malloc_lock = &malloc_lock_nop;
      |                      ^~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:262:5: error: called object 'malloc_unlock' is not a function or function pointer
  262 |     malloc_unlock();
      |     ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:31:24: note: declared here
   31 | static malloc_unlock_t malloc_unlock = &malloc_unlock_nop;
      |                        ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c: At top level:
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:265:25: error: unknown type name 'malloc_lock_t'
  265 | void set_malloc_locking(malloc_lock_t lock, malloc_unlock_t unlock)
      |                         ^~~~~~~~~~~~~
repos/apache-mynewt-core/libc/baselibc/src/malloc.c:265:45: error: unknown type name 'malloc_unlock_t'
  265 | void set_malloc_locking(malloc_lock_t lock, malloc_unlock_t unlock)
      |                                             ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```
And thank you for your patience @sjanc .

sjanc, 2021-04-27:
that won't work, you mix newlib headers and baselibc

if you want to use newlib you'd have to hack a bit, eg bsp are pulling baselibc package

sjanc, 2021-04-28:
btw I think you should be able to provide PRIu64 definition from cflags just like you tried with newlib include path eg  -DPRIu64="llu"

danielkucera, 2021-04-28:
Okay, this has worked:
```
newt target set nrf52_blinky cflags='-DPRIu64="llu" -DPRIx64="llx"'
```
I can live with that.

sjanc, 2021-04-28:
great, I'd suggest to fill bug against ubuntu arm gcc too

sjanc, 2021-05-25:
please re-open if needed

closed 2021-05-25 by sjanc