apache/mynewt-core #2623: IPC_NRF5340_NET_GPIO doesn't allow for more than a single pin value (?)

ncasaril, 2021-06-23:
The MYNEW_VAL(IPC_NRF5340_NET_GPIO) is used below both as an expression to the preprocessor ```#if``` statement and as the initialiser for the gpios vector here:

https://github.com/apache/mynewt-core/blob/8a853d0e841b48259f8c5e8c77e0cb29fa88f03c/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c#L164-L167

Which, from what I can see, only works if IPC_NRF5340_NET_GPIO is a single pin. It fails for any of the below attempts of using more than one value in syscfg.vals:

```IPC_NRF5340_NET_GPIO: 28, 29```

```
Error: In file included from repos/apache-mynewt-core/kernel/os/include/os/os.h:26,
                 from repos/apache-mynewt-core/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c:21:
repos/apache-mynewt-core/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c: In function 'ipc_nrf5340_init':
bin/targets/pca10095_blinky/generated/include/syscfg/syscfg.h:190:44: error: left-hand operand of comma expression has no effect [-Werror=unused-value]
 #define MYNEWT_VAL_IPC_NRF5340_NET_GPIO (28, 29)
                                            ^
bin/targets/pca10095_blinky/generated/include/syscfg/syscfg.h:15:49: note: in expansion of macro 'MYNEWT_VAL_IPC_NRF5340_NET_GPIO'
 #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
                                                 ^~~~~~~~~~~
repos/apache-mynewt-core/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c:165:30: note: in expansion of macro 'MYNEWT_VAL'
     unsigned int gpios[] = { MYNEWT_VAL(IPC_NRF5340_NET_GPIO) };
```

```IPC_NRF5340_NET_GPIO: '"28, 29"'```

```
Error: In file included from repos/apache-mynewt-core/kernel/os/include/os/os.h:26,
                 from repos/apache-mynewt-core/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c:21:
bin/targets/pca10095_blinky/generated/include/syscfg/syscfg.h:190:42: error: token ""28, 29"" is not valid in preprocessor expressions
 #define MYNEWT_VAL_IPC_NRF5340_NET_GPIO ("28, 29")
                                          ^~~~~~~~
bin/targets/pca10095_blinky/generated/include/syscfg/syscfg.h:15:49: note: in expansion of macro 'MYNEWT_VAL_IPC_NRF5340_NET_GPIO'
 #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
                                                 ^~~~~~~~~
repos/apache-mynewt-core/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c:24:5: note: in expansion of macro 'MYNEWT_VAL'
 #if MYNEWT_VAL(IPC_NRF5340_NET_GPIO)
     ^~~~~~~~
repos/apache-mynewt-core/hw/drivers/ipc_nrf5340/src/ipc_nrf5340.c: In function 'ipc_nrf5340_init':
bin/targets/pca10095_blinky/generated/include/syscfg/syscfg.h:190:42: error: token ""28, 29"" is not valid in preprocessor expressions
 #define MYNEWT_VAL_IPC_NRF5340_NET_GPIO ("28, 29")
```

sjanc, 2021-06-23:
(slightly over complicated) fix fot this 

https://github.com/apache/mynewt-core/pull/2628

ncasaril, 2021-06-27:
Solved by #2629.

closed 2021-06-27 by ncasaril