apache/mynewt-core #2704: TUD_DFU_DESC_LEN is never defined

t3zeng, 2021-10-29:
There is a calculation for `CONFIG_TOTAL_LEN` inside `apache-mynewt-core/hw/usb/tinyusb/std_descriptors/src/std_descriptors.c` that makes use of `TUD_DFU_DESC_LEN` which doesn't seem to ever get defined anywhere.

t3zeng, 2021-10-29:
`
#define CONFIG_TOTAL_LEN    (TUD_CONFIG_DESC_LEN + \
                             CFG_TUD_CDC * TUD_CDC_DESC_LEN + \
                             CFG_TUD_MSC * TUD_MSC_DESC_LEN + \
                             CFG_TUD_HID * TUD_HID_DESC_LEN + \
                             CFG_TUD_BTH * TUD_BTH_DESC_LEN + \
                             CFG_TUD_DFU * TUD_DFU_DESC_LEN(1) + \
                             0)`

kasjer, 2021-11-03:
It is defined in usbd.h

t3zeng, 2021-11-03:
Yep looks like that was my issue. Thanks!

closed 2021-11-03 by t3zeng