apache/mynewt-core #2754: Error in semihost_rename() function

kemonats, 2022-01-24:
The `semihost_rename()` function is currently:
https://github.com/apache/mynewt-core/blob/12ee28f2052b7c716cdf8e3ac49f52f7c372f331/hw/drivers/semihosting/src/mbed_semihost_api.c#L103-L111
and should be:
```
int semihost_rename(const char *old_name, const char *new_name)
{
    uint32_t args[4];
    args[0] = (uint32_t)old_name;
    args[1] = (uint32_t)strlen(old_name);
    args[2] = (uint32_t)new_name;
    args[3] = (uint32_t)strlen(new_name);
    return __semihost(SYS_RENAME, args);
}
```
[Semihosting for AArch32 and AArch64](https://github.com/ARM-software/abi-aa/blob/main/semihosting/semihosting.rst#sys-rename-0x0f)

## 6.16   SYS_RENAME (0x0F)
Renames a specified file.

### 6.16.1   Entry
On entry, the PARAMETER REGISTER contains a pointer to a four-field data block:

### field 1
- A pointer to the name of the old file.
### field 2
- The length of the old filename.
### field 3
- A pointer to the new filename.
### field 4
- The length of the new filename.

Both strings are null-terminated.

### 6.16.2   Return
On exit, the RETURN REGISTER contains:

- 0 if the rename is successful.
- A nonzero, host-specific error code if the rename fails.

sjanc, 2022-01-25:
Hello,

looks like typo indeed,  @kemonats do you want to send PR with this fix?:)

kemonats, 2022-01-25:
No

closed 2022-04-13 by sjanc