apache/mynewt-core #2857: Is there a .pdf documentation for  apache-mynewt?

ujur007, 2022-07-03:
I am working with Apache mynewt. It has quite good documentation but it is only online and moving from one tab to another is hectic.

Do you know is there a pdf version for the documentation?

sjanc, 2022-09-13:
I don't think we have PDF documentation unfortunatelly

sjanc, 2024-08-22:
@wpiet  please have a look at this in spare time

sjanc, 2024-09-27:
Hi,

There is initial (experimental) PDF documentation available at https://mynewt.apache.org/documentation/

Currently there is one only for master branch, but we plan to drop it on next release and just keep with PDF documentation for released versions

closed 2024-09-27 by sjanc

ujur007, 2024-09-27:
Thanks. Appreciate your efforts as it will help me to navigate through the API easily.