apache/mynewt-core #2944: Enabling uart 1 on nrf52dk?

suadsuljic7, 2023-02-06:
Hi I m trying to use uart 1 on nrf52dk pca10056. In the syscfg.yml i add syscfg.vals:
    UART_1: 1
    UART_1_PIN_TX: 15
    UART_1_PIN_RX: 14
    UART_1_PIN_RTS: 5
    UART_1_PIN_CTS: 7
and I'm using the hal_uart_blocking_tx() function to send the data. But I receive nothing on the serial.

kasjer, 2023-09-21:
With syscfg that you have try this code:
```c
    struct uart_conf uc = {
            .uc_speed = 115200,
            .uc_databits = 8,
            .uc_stopbits = 1,
            .uc_parity = UART_PARITY_NONE,
            .uc_flow_ctl = UART_FLOW_CTL_NONE,
            .uc_tx_char = NULL,
            .uc_rx_char = NULL,
    };
    struct uart_dev *uart = uart_open("uart1", &uc);
    uart_blocking_tx(uart, 'h');
```

using hal_uart_blocking_tx() requires more configuration that is done here partially by syscfg `UART_1: 1` and by `uart_open()`

sjanc, 2024-09-06:
please reopen if this is stil an issue

closed 2024-09-06 by sjanc