apache/mynewt-core #3064: apache-mynewt-core repo dirty state and Blinky sample nrf52_boot build failure

RayCxggg, 2023-09-13:
Hi, I'm building Blinky sample for nRF52 board according to the [doc](https://mynewt.apache.org/latest/tutorials/blinky/nRF52.html#load-the-bootloader-and-the-blinky-application-image). I'm using Ubuntu 20.04 in WSL.
When I ran `newt upgrade`, the error shows:
```
[apache-mynewt-core]:
Downloading repository description
Fetching repo mynewt-core
Download of "repository.yml" from repo:apache-mynewt-core commit:master successful
Error: some repos are in a dirty state:
    apache-mynewt-core: contains local changes
Specify the `-f` (force) switch to attempt anyway
```

I then tried `newt upgrade -f`, but got:
```
WARNING: some repos are in a dirty state:
    apache-mynewt-core: contains local changes

Making the following changes to the project:
    upgrade apache-mynewt-core (0.0.1 --> 1.9.0)
    upgrade apache-mynewt-mcumgr (0.4.0 --> 0.2.0)
    upgrade apache-mynewt-nimble (3a161c8e3e50bbf201fbebcb92558240eaf71aaf --> 1.4.0)
    upgrade mcuboot (0.0.0 --> 1.7.2)
Error: Error updating "apache-mynewt-core": error: Your local changes to the following files would be overwritten by checkout:
        .github/project.yml
        .github/targets/da1469x_flash_loader/pkg.yml
        .github/targets/da1469x_flash_loader/syscfg.yml
        .github/targets/da1469x_flash_loader/target.yml
        .github/targets/nordic_pca10056_advertiser/pkg.yml
        .github/targets/nordic_pca10056_advertiser/target.yml
        .github/targets/nordic_pca10056_blecent/pkg.yml
        .github/targets/nordic_pca10056_blecent/target.yml
        .github/targets/nordic_pca10056_blecsc/pkg.yml
        .github/targets/nordic_pca10056_blecsc/target.yml
        .github/targets/nordic_pca10056_blehci/pkg.yml
        .github/targets/nordic_pca10056_blehci/target.yml
        .github/targets/nordic_pca10056_blehci_strict_sched/pkg.yml
        .github/targets/nordic_pca10056_blehci_strict_sched/syscfg.yml
        .github/targets/nordic_pca10056_blehci_strict_sched/target.yml
        .github/targets/nordic_pca10056_blehr/pkg.yml
        .github/targets/nordic_pca10056_blehr/target.yml
        .github/targets/nordic_pca10056_blemesh/pkg.yml
        .github/targets/nordic_pca10056_blemesh/target.yml
        .github/targets/nordic_pca10056_blemesh_cdb/pkg.yml
        .github/targets/nordic_pca10056_blemesh_cdb/syscfg.yml
        .github/targets/nordic_pca10056_blemesh_cdb/target.yml
        .github/targets/nordic_pca10056_blemesh_ext_adv/pkg.yml
        .github/targets/nordic_pca10056_blemesh_ext_adv/syscfg.yml
        .github/targets/nordic_pca10056_blemesh_ext_adv/target.yml
        .github/targets/nordic_pca10056_blemesh_light/pkg.yml
        .github/targets/nordic_pca10056_blemesh_light/target.yml
        .github/targets/nordic_pca10056_blemesh_models_example_1/pkg.yml
        .github/targets/nordic_pca10056_blemesh_models_example_1/target.yml
        .github/targets/nordic_pca10056_blemesh_models_example_2/pkg.yml
        .github/targets/nordic_pca10056_blemesh_models_example_2/target.yml
        .github/targets/nordic_pca10056_blemesh_shell/pkg.yml
        .github/targets/nordic_pca10056_blemesh_shell/target.yml
        .github/targets/nordic_pca10056_bleprph/pkg.yml
        .github/targets/nordic_pca10056_bleprph/target.yml
        .github/targets/nordic_pca10056_bleprph_oic/pkg.yml
        .github/targets/nordic_pca10056_bleprph_oic/target.yml
        .github/targets/nordic_pca10056_blesplit/pkg.yml
        .github/targets/nordic_pca10056_blesplit/target.yml
        .github/targets/nordic_pca10056_bleuart/pkg.yml
        .github/targets/nordic_pca10056_bleuart/target.yml
        .github/targets/nordic_pca10056_boot/pkg.yml
        .github/targets/nordic_pca10056_boot/target.yml
        .github/targets/nordic_pca10056_btshell/pkg.yml
        .github/targets/nordic_pca10056_btshell/target.yml
        .github/targets/nordic_pca10056_ext_advertiser/pkg.yml
        .github/targets/nordic_pca10056_ext_advertiser/target.yml
        .github/targets/nordic_pca10056_ocf_sample/pkg.yml
        .github/targets/nordic_pca10056_ocf_sample/target.yml
        .github/targets/nordic_pca10056_pwm_test/pkg.yml
        .github/targets/nordic_pca10056_pwm_test/syscfg.yml
        .github/targets/nordic_pca10056_pwm_test/target.yml
        .github/targets/nordic_pca10056_scanner/pkg.yml
        .github/targets/nordic_pca10056_scanner/target.yml
        .github/targets/nordic_pca10056_sensors_test/pkg.yml
        .github/targets/nordic_pca10056_sensors_test/target.yml
        .github/targets/nordic_pca10056_slinky/pkg.yml
        .github/targets/nordic_pca10056_slinky/target.yml
        .github/targets/nordic_pca10056_slinky_oic/pkg.yml
        .github/targets/nordic_pca10056_slinky_oic/target.yml
        .github/targets/nordic_pca10056_spitest/pkg.yml
        .github/targets/nordic_pca10056_spitest/target.yml
        .github/targets/nordic_pca10056_splitty/pkg.yml
        .github/targets/nordic_pca10056_splitty/target.yml
        .github/targets/nordic_pca10056_testbench/pkg.yml
        .github/targets/nordic_pca10056_testbench/target.yml
        .github/targets/nordic_pca10056_timtest/pkg.yml
        .github/targets/nordic_pca10056_timtest/target.yml
        .github/targets/nordic_pca10056_trng_test/pkg.yml
        .github/targets/nordic_pca10056_trng_test/target.yml
        .github/targets/nordic_pca10095_blehci/pkg.yml
        .github/targets/nordic_pca10095_blehci/syscfg.yml
        .github/targets/nordic_pca10095_blehci/target.yml
        .github/targets/nordic_pca10095_btshell/pkg.yml
        .github/targets/nordic_pca10095_btshell/syscfg.yml
        .github/targets/nordic_pca10095_btshell/targe
Aborting
```
I managed to find a [closed issue](https://github.com/apache/mynewt-core/issues/2507#issuecomment-788858842) in 2021 which is similar, but it doesn't work in my condition. However, I'm able to build the [Blinky sample](https://mynewt.apache.org/latest/get_started/project_create.html) and run in simulator, so I temporarily moved on to the Blinky project on nRF52 board.

However, when I ran `newt build nrf52_boot`, the error shows:
```
Building target targets/nrf52_boot
Error: Config for package @apache-mynewt-core/hw/mcu/nordic/nrf52xxx: setting MCU_ACCESS_PORT_PROTECTION has invalid choice defined (enable,disable,default) - only letters, numbers and underscore are allowed
```

How can I solve the problem? Thank you.

sjanc, 2023-09-13:
Hi,

Try clearing up changes in core repo before retrying (cd repos/apache-mynewt-core; git stash) and retrying

I'd also suggest to try with newt 1.11 (even if you want to use Mynewt 1.9) as there were some fixes in handling repositories

RayCxggg, 2023-09-13:
@sjanc The suggestions are very helpful, thank you. 

Since I'm building Mynewt with WSL on Windows, I can't directly flash the firmware from WSL with `newt load`. Is there a way that I can use tools on Windows like `nrfjprog` to simply flash the HEX file?

closed 2023-09-14 by RayCxggg

RayCxggg, 2023-09-15:
I figure that with [usbipd](https://learn.microsoft.com/en-us/windows/wsl/connect-usb), WSL2 is able to access Windows USB devices as real hardware from Linux side. Then, I can directly use `newt load` to flash firmware in WSL2.

I will also try the [nrfjprog.sh](https://github.com/apache/mynewt-core/blob/master/hw/scripts/nrfjprog.sh) script. Thank you!

5614037u, 2023-09-29:
> Hi,
> 
> Try clearing up changes in core repo before retrying (cd repos/apache-mynewt-core; git stash) and retrying
> 
> I'd also suggest to try with newt 1.11 (even if you want to use Mynewt 1.9) as there were some fixes in handling repositories

Yeah I gave this a go but it does not fix the issue. Is this issue linked with WSL?

5614037u, 2023-09-30:
> > Hi,
> > Try clearing up changes in core repo before retrying (cd repos/apache-mynewt-core; git stash) and retrying
> > I'd also suggest to try with newt 1.11 (even if you want to use Mynewt 1.9) as there were some fixes in handling repositories
> 
> Yeah I gave this a go but it does not fix the issue. Is this issue linked with WSL? I also used 1.11

Ok I got it to work but I had to use a native WSL folder e.g ~/dev why does running in /mnt/ break it?