apache/mynewt-core #3066: RAM overflow error when build app for PineTime

Kisszalimo, 2023-09-13:
Hi,
I would like to try out a [simple demo app](https://github.com/apache/mynewt-core/pull/2951/commits/0ad5f0ed5d452dafca8c56ed45f2682ebc94c602) using LVGL which was introduced in Mynewt v1.11. Unfortunately I get an error while building project:

```
$ newt build limon_v1
...
Error: /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/(...)/apache-mynewt-newt-bin-linux-1.11.0/limon_v1/bin/targets/limon_v1/app/apps/blinky/blinky.elf section `.bss' will not fit in region `RAM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 5156 bytes
collect2: error: ld returned 1 exit status
```

If I remember correctly, I only added ```@apache-mynewt-core/hw/drivers/display/lvgl``` into app ```pkg.yml``` as a package dependency and set LVGL display resolution and ```LCD_DC_PIN``` in the appropriate ```syscfg.yml``` file. I did not modify any further files and I use built-in LVGL demos.

kasjer, 2023-09-14:
@Kisszalimo, while I don't have pinetime watch to verify that it works, here is what you may try:

There is pending PR #3063 that adds options that can tweak LVGL a little bit so it will fit in memory.
You could merge commits from this PR to your mynewt-core branch.
I hope PR will be merged soon.

Next you can change you target syscfg.yml to have this:
```yml
syscfg.vals:
    LCD_ITF: spi
    LCD_SPI_DEV_NAME: '"spi0"'
    LCD_SPI_FREQ: 8000
    LCD_SPI_MODE: BUS_SPI_MODE_0
    LCD_CS_PIN: 25
    LCD_RESET_PIN: 26
    LCD_DC_PIN: 18
    LCD_BL_PIN: 13
    LVGL_DISPLAY_HORIZONTAL_RESOLUTION: 240
    LVGL_DISPLAY_VERTICAL_RESOLUTION: 240
    LVGL_ST7789: 1
   # this reduce memory usage of LVGL
    LV_MEM_SIZE: 40000
```
I did test it on **nordic_pca10040** target that has same MCU but no other peripherals except LCD

Kisszalimo, 2023-09-17:
@kasjer thank you for your response. I added a new repo including the PR you mentioned to my project. On the first try, the problem was not solved:

```
 Error: /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/(...)/apache-mynewt-newt-bin-linux-1.11.0/limon_v1/bin/targets/limon_v1/app/apps/blinky/blinky.elf section `.bss' will not fit in region `RAM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 248 bytes
```
However, the amount of overflowed bytes has decreased a lot. So on the second try, I modified my target `syscfg.yml` file according to your advice, and then the project built successfully.

Before Mynewt supported LVGL, there was [an unofficial port](https://github.com/lupyuen/pinetime_lvgl_mynewt/tree/master) to Mynewt. The main difference is the version of LVGL: the unofficial port uses V7, while Mynewt port uses V8. I thought it would be interesting to compare these two ports, so I ran a [benchmark demo](https://github.com/lvgl/lvgl/tree/master/demos/benchmark) to test both ports. Unfortunatelly it seems the older port cannot measure the performance properly (or I did something wrong), because it shows very high results (e.g. weighted FPS: 9000), which does not make any sense.

So finally I created two very simple scrolling-label demos using both ports. The functionality and included libraries (except ports) are the same. Strangely the size of generated image files differs significantly:

**Image size using Mynewt port:** 219KB
**Image size using older port:** 138KB

I am wondering what causes this big difference and how can we reduce it. Probably the new version of LVGL requires more space... or newt wants to compile "too much thing" into the image?

kasjer, 2023-09-20:
@Kisszalimo, I checked some more, and it looks like LVGL has now more features that take space.
Here is a way to trim your build.
In your application or target **syscfg.yml** add line
```yml
LVGL_DEFAULT_LV_CONF: 0
```
This will prevent usage of mynewt provided `lv_conf.h` file which can customize some of LVGL settings from mynewt-like **syscfg.yml** files but it does not cover most of them, only some basic configuration that is ok to run LVGL demos.

Copy `hw/drivers/display/lvgl/default/lv_conf.h` file to your application include directory, and disable all LVGL features that are not needed for your scrolling-label demo (maybe calendar, arc and such)

If you want to execute LVGL demos that are covered by **apps/lvgl/demo** with your customized **lv_conf.h** you can add **include** directory to folder where your target is located (i.e. `targets/pinetime-lvgl_demo/pkg.yml`) add line
```yml
pkg.cflags:
    - -Iinclude
```
That will add include to all compiled files and then you can have your lv_conf.h that can be fully customized with all the features that LVGL exposes.

Kisszalimo, 2023-09-25:
@kasjer i was able to reduce my build by applying your advice. Now it uses ~180KB space which is a notable difference (of course it depends on the needs of your application). However if flash memory is a bottleneck, maybe you should consider using an older verson of LVGL.

I am fairly new to mynewt and sometimes the project structure and its working mechanism is a bit mysterious for me. For example, if I do not add
```
LVGL_DEFAULT_LV_CONF: 0
```
to my app `syscfg.yml` (leave it default), it seems mynewt use my custom `lv_conf.h` anyway, instead of ignoring it and use the default LVGL config file as expected. The only way to force mynewt to use default config file is to delete my custom config file from the project.

kasjer, 2023-09-26:
For now there is no effort to have older version supported.
If I have time I will see if we can override LVGL version in project.yml (I'm not sure if this will require some work on newt tool though).

You are right about lv_conf.h if you have one added to application or target it will automatically override any file with the same name in the project. For now this is the case because we include it by `#inlude <lv_conf.h>`. LVGL have more options to specify config file that I may try out once all things are settled.

I'm not expert on LVGL by any means so if you want to contribute to mynewt project in this area your contribution will be appreciated.

closed 2023-09-26 by Kisszalimo