apache/mynewt-core #3070: NimBLE sample build error with apache-mynewt-core

RayCxggg, 2023-09-16:
Hi,

I'm trying to build the BLE sample following the [Set up a bare bones NimBLE application](https://mynewt.apache.org/latest/tutorials/ble/ble_bare_bones.html). I'm working on Ubuntu 20.04 with WSL2.

There is one step in the doc to add following configuration in syscfg.yml:
```
syscfg.vals:
    BLE_HCI_TRANSPORT: builtin
```

When I ran `newt build ble_tgt`, the following error appeared:
```
Building target targets/ble_tgt
Error: Override of defunct settings detected:
    BLE_HCI_TRANSPORT: see doc/transport.md

Setting history (newest -> oldest):
    BLE_HCI_TRANSPORT: [apps/ble_app:builtin, @apache-mynewt-nimble/nimble/transport:]
```
It seems that the `BLE_HCI_TRANSPORT` has been removed(the tutorial is outdated), but I can't find where `doc/transport.md` is.

So I temporarily removed the config in syscfg.yml, but then I got another error while building:
```
Error: repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c: In function 'value_to_pretty':
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:334:33: error: expected ')' before 'PRIu64'
  334 |             if (fprintf(out, "%" PRIu64, val) < 0)
      |                                 ^~~~~~~
      |                                 )
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:334:31: error: spurious trailing '%' in format [-Werror=format=]
  334 |             if (fprintf(out, "%" PRIu64, val) < 0)
      |                               ^
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:343:38: error: expected ')' before 'PRIu64'
  343 |                 if (fprintf(out, "-%" PRIu64, val) < 0)
      |                                      ^~~~~~~
      |                                      )
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:343:36: error: spurious trailing '%' in format [-Werror=format=]
  343 |                 if (fprintf(out, "-%" PRIu64, val) < 0)
      |                                    ^
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:390:29: error: expected ')' before 'PRIu64'
  390 |         if (fprintf(out, "%" PRIu64 "(", tag) < 0)
      |                             ^~~~~~~
      |                             )
repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c:390:27: error: spurious trailing '%' in format [-Werror=format=]
  390 |         if (fprintf(out, "%" PRIu64 "(", tag) < 0)
      |                           ^
cc1: all warnings being treated as errors
```
It seems to be a C standard problem which is related to the compiler. The arm-none-eabi-gcc version I'm using is 9.2.1, and I don't know which version is required. The [doc](https://mynewt.apache.org/latest/get_started/native_install/cross_tools.html#installing-the-arm-toolchain-for-linux) suggests a 4.9.3 version gcc. Should I downgrade to an old version?

RayCxggg, 2023-09-24:
[Solutions ](https://stackoverflow.com/questions/8132399/how-to-printf-uint64-t-fails-with-spurious-trailing-in-format) I can find all suggest that this is a C99 standard problem. 

I have add below code to `repos/apache-mynewt-core/encoding/tinycbor/src/cborpretty.c` and `repos/apache-mynewt-core/encoding/tinycbor/src/cbortojson.c`:
```
#define __STDC_FORMAT_MACROS 1
...
#include <inttypes.h>
```
But it doesn't solve the problem. My compiler version is `arm-none-eabi-g++ (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]`, what is the suggesting version of gcc to build MynewtOS?

I find no `__STDC_FORMAT_MACROS` in the entire Mynewt repo. Does that mean such error has never occurred before?

sjanc, 2023-09-24:
I think this is issue with ubuntu (debian) packaging, try with toolchain from ARM

sjanc, 2023-09-25:
for the record, if you would like to push this on ubuntu, there is a debian issue https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=953844 which seems to be what you hit on ubuntu

RayCxggg, 2023-09-26:
Thanks a lot. I will try to report to Ubuntu community.