apache/mynewt-core #3106: Connecting to a Xiaomi Flower Care

jaaseru, 2023-11-13:
I am using an ESP32 with NimBLE, and I want to connect to a Xiaomi Flower care sensor. I have managed to scan and pair with the sensor while using Bluedroid. I can also scan for the sensor and find its advertisement packets using NimBLE. However I am having issues pairing. Upon calling ble_gap_connect() I get a timeout error in the event handler. I suspect this is because my connection parameters are wrong:

    ```struct ble_gap_conn_params conn_params;
    conn_params.scan_itvl = 0x0010;
    conn_params.scan_window = 0x0010;
    conn_params.itvl_min = 0x0006;
    conn_params.itvl_max = 0x0006;
    conn_params.latency = 0;
    conn_params.supervision_timeout = 1000;
    conn_params.min_ce_len = 0x0000;
    conn_params.max_ce_len = 0x0000;```
    
I'm not sure what else I could tune in the connect-function or event handler to get it to connect. The Sensor is definitely present and connectable. Any advice on the connection parameters is appreciated.

sjanc, 2023-11-21:
Hi,

I'd first check with some wider range of connection intervals, eg set conn_params.itvl_max to 40 for 50ms max conn interval.   Also check which error code you get,  ie it is 'failed to be establish' or 'supervision timeout', this may help in debugging issue

And note that ESP32 is using only NimBLE host while controller is from ESP so if above doesn't help you may need to ask ESP32 folks as connection creation is controller's job...

sjanc, 2024-03-06:
is this stil an issue?

jaaseru, 2024-03-06:
Hi! I managed to resolve this and continue the work. The timeout I suspect came from user error when trying to use the NimBLE API. The connection parameters had little to dowith it. It also seems to be a power saving feature of a sort that the flower care disconnects automatically after a few seconds of open connection.

closed 2024-03-06 by sjanc