apache/mynewt-core #3109: Linker error: undefined reference to `console_rx_restart' when using Mynewt Console API

Kisszalimo, 2023-11-20:
I would like to achieve some basic I/O operations with console. I can use `console_write()` successfully to print messages on console but it would be great if i could read characters from it. I tried `console_read()` but I got a linker error:

```
Error: /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/.../consy-pinetime/bin/targets/blinky-pinetime/app/@apache-mynewt-core/sys/console/full/@apache-mynewt-core_sys_console_full.a(console.o): in function `console_line_event_put':
/home/.../consy-pinetime/repos/apache-mynewt-core/sys/console/full/src/console.c:1274: undefined reference to `console_rx_restart'
collect2: error: ld returned 1 exit status
```

Minimal code example:

```
#include <string.h>
#include <console/console.h>

int dl_semihost_read(char *buff) {
    return console_read(buff, strlen(buff), NULL);
}
```

sjanc, 2023-11-21:
you need to include console package in your app (or target)
eg put this in your targets pkg.yml file:
pkg.deps:
    - "@apache-mynewt-core/sys/console"

After that you can tune console via syscfg.yml (by default full UART console is enabled), check repos/apache-mynewt-core/sys/console/full/syscfg.yml for details

Kisszalimo, 2023-11-21:
I already included it in my app:

```
pkg.name: apps/blinky
pkg.type: app
pkg.description: Basic example application which blinks an LED.
pkg.author: "Apache Mynewt <dev@mynewt.apache.org>"
pkg.homepage: "http://mynewt.apache.org/"
pkg.keywords:

pkg.deps:
    - "@apache-mynewt-core/kernel/os"
    - "@apache-mynewt-core/hw/hal"
    - "@apache-mynewt-core/sys/console/full"
    - "@apache-mynewt-core/sys/log/stub"
```

I used `@apache-mynewt-core/sys/console/full` instead of `@apache-mynewt-core/sys/console` but as I see it, it doesn't matter which package I use (or both), I get the same linker error anyway.

sjanc, 2023-11-22:
can you post "newt target revdep YOUR_TARGET" ?

Kisszalimo, 2023-11-22:
Sure, here it is:

```
$ newt target revdep blinky-pinetime
Reverse dependency graph (dependee <-- [dependers]):
    * @apache-mynewt-core/compiler/arm-none-eabi-m4 <-- []
    * @apache-mynewt-core/hw/battery <-- [@apache-mynewt-core/hw/bsp/pinetime(syscfg:BSP_BATTERY)]
    * @apache-mynewt-core/hw/bsp/pinetime <-- []
    * @apache-mynewt-core/hw/charge-control <-- [@apache-mynewt-core/hw/drivers/chg_ctrl/sgm4056]
    * @apache-mynewt-core/hw/cmsis-core <-- [@apache-mynewt-core/hw/drivers/semihosting @apache-mynewt-core/hw/mcu/nordic @apache-mynewt-core/hw/mcu/nordic/nrf52xxx]
    * @apache-mynewt-core/hw/drivers/adc <-- [@apache-mynewt-core/hw/bsp/pinetime(syscfg:BSP_BATTERY) @apache-mynewt-core/hw/drivers/adc/adc_nrf52]
    * @apache-mynewt-core/hw/drivers/adc/adc_nrf52 <-- [@apache-mynewt-core/hw/mcu/nordic/nrf52xxx(syscfg:ADC_0)]
    * @apache-mynewt-core/hw/drivers/chg_ctrl/sgm4056 <-- [@apache-mynewt-core/hw/bsp/pinetime(syscfg:BSP_CHARGER)]
    * @apache-mynewt-core/hw/drivers/flash/spiflash <-- [@apache-mynewt-core/hw/bsp/pinetime]
    * @apache-mynewt-core/hw/drivers/flash/spiflash/chips <-- [@apache-mynewt-core/hw/drivers/flash/spiflash]
    * @apache-mynewt-core/hw/drivers/semihosting <-- [@apache-mynewt-core/sys/console/full(syscfg:CONSOLE_SEMIHOSTING)]
    * @apache-mynewt-core/hw/hal <-- [@apache-mynewt-core/hw/drivers/chg_ctrl/sgm4056 @apache-mynewt-core/hw/drivers/flash/spiflash @apache-mynewt-core/hw/mcu/nordic @apache-mynewt-core/hw/mcu/nordic/nrf52xxx @apache-mynewt-core/sys/console/full apps/blinky]
    * @apache-mynewt-core/hw/mcu/nordic <-- [@apache-mynewt-core/hw/drivers/adc/adc_nrf52 @apache-mynewt-core/hw/mcu/nordic/nrf52xxx]
    * @apache-mynewt-core/hw/mcu/nordic/nrf52xxx <-- [@apache-mynewt-core/hw/bsp/pinetime]
    * @apache-mynewt-core/kernel/os <-- [@apache-mynewt-core/hw/battery @apache-mynewt-core/hw/bsp/pinetime @apache-mynewt-core/hw/charge-control @apache-mynewt-core/hw/drivers/chg_ctrl/sgm4056 @apache-mynewt-core/hw/hal @apache-mynewt-core/libc/baselibc @apache-mynewt-core/sys/console/full @apache-mynewt-core/sys/log/modlog @apache-mynewt-core/sys/sysdown @apache-mynewt-core/sys/sysinit @apache-mynewt-core/util/mem @apache-mynewt-core/util/rwlock apps/blinky]
    * @apache-mynewt-core/libc/baselibc <-- [@apache-mynewt-core/hw/bsp/pinetime]
    * @apache-mynewt-core/sys/console <-- [apps/blinky targets/blinky-pinetime]
    * @apache-mynewt-core/sys/console/full <-- [@apache-mynewt-core/hw/battery(api:console) @apache-mynewt-core/hw/charge-control(api:console) @apache-mynewt-core/kernel/os(api:console) @apache-mynewt-core/libc/baselibc(api:console) apps/blinky]
    * @apache-mynewt-core/sys/defs <-- [@apache-mynewt-core/sys/flash_map]
    * @apache-mynewt-core/sys/flash_map <-- [@apache-mynewt-core/sys/sysinit]
    * @apache-mynewt-core/sys/log/common <-- [@apache-mynewt-core/sys/log/stub]
    * @apache-mynewt-core/sys/log/modlog <-- [@apache-mynewt-core/sys/flash_map]
    * @apache-mynewt-core/sys/log/stub <-- [@apache-mynewt-core/sys/log/modlog(api:log) apps/blinky]
    * @apache-mynewt-core/sys/sys <-- [@apache-mynewt-core/kernel/os]
    * @apache-mynewt-core/sys/sysdown <-- [@apache-mynewt-core/kernel/os]
    * @apache-mynewt-core/sys/sysinit <-- [@apache-mynewt-core/kernel/os]
    * @apache-mynewt-core/util/mem <-- [@apache-mynewt-core/kernel/os]
    * @apache-mynewt-core/util/rwlock <-- [@apache-mynewt-core/sys/log/modlog]
    * apps/blinky <-- []
    * targets/blinky-pinetime <-- []
```

sjanc, 2023-11-23:
Hi,

OK, I've miss that before but you are using pinetime, and it is using semihosting console which currently has only TX implemented

Kisszalimo, 2023-11-23:
Thank you for your help. Yes, pinetime using arm semihosting. It has a 4 pin SWD interface. 

Hmm... this is unfortunate, because it would be an easy way to communicate with the device. Will it ever be implemented in the future?

andrzej-kaczmarek, 2023-11-23:
use RTT console instead

Kisszalimo, 2023-11-25:
Yes, that could be another option, but as I know, RTT is only available for SEGGER debuggers and I have a simple ST-Link v2 programmer.

closed 2023-12-16 by Kisszalimo