apache/mynewt-core #3110: BUILD fail for nrf5340dk board Error: Could not resolve BSP package: @apache-mynewt-core/hw/bsp/nordic_pca10095

SatishSolankeEsp, 2023-11-21:
Hi 
I am trying to create the ble project with blehci , created two target for nrf5340dk
[image]
but show the above error while building the target.

i could also see the couple of warning as below :
[image]

Please help to build project for nrf5340dk. I would like flash the mynewt ble controller on nrf via hci .

Thanks,
Satish

sjanc, 2023-11-21:
Hi,

this looks like repositories are not installed,  did you run 'newt upgrade' ?  what does 'newt info' shows?

SatishSolankeEsp, 2023-11-21:
[image]
[image]
newt info and newt upgrade shows warning like above .

sjanc, 2023-11-22:
looks like you are using very old version (1.4) of mynewt,  please get latest newt tool from https://mynewt.apache.org/download/ and than either create new project or adjust your project.yml to point to 1.9.0 and do newt upgrade

SatishSolankeEsp, 2023-11-23:
[image]
I Have change newt to 1.10.0 and still not able to build shows above errors .

I am not able to create new project show below error 
[image]

sjanc, 2023-11-23:
newt 1.10 had a bug, so either use newt 1.11 or try 'newt new --shallow=0 sniffproj'

as for unknown gcc option, you probably need to upgrade your toolchain

SatishSolankeEsp, 2023-11-23:
I switch back to newt 1.9.0 works fine able to create new proj.
Build shows new error now:
[image]

for gcc error i will check how to upgrade toolchain
[image]

sjanc, 2023-11-23:
I strongly suggest to test with newt 1.11, there were a lot of nrf53 related improvements in latest releases

As for ARM GCC update, you can either update ubuntu (18.04 is quite old...) or download one from ARm directly https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads   (I suggest using GCC 12 as 13 may give you some compilation warning we haven't fix yet)

Also, you may want to check cwiki for nRF53 on Mynewt: https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=177047756   (and I'd recommend to use NRF5340_EMBED_NET_CORE:1 option instead of building own netcore image)

sjanc, 2024-03-06:
Hi,  is this stil an issue?

closed 2024-03-06 by sjanc