apache/mynewt-core #3310: Log module throws error when trying to build basic ble advertising application.

ujur007, 2024-09-23:
I tried to build a simple ble advertiser application in apps/advertiser and got the error related to implicit declaration for the function and type cast of a variable. 
Here is what my target looks like.

```
targets/blues
    app=@apache-mynewt-nimble/apps/advertiser
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=debug

```

The error when trying to build the application is as follows. For me the application worked when I did some hacks by commenting the code that throws error.

```
Error: repos/apache-mynewt-core/sys/log/full/src/log.c: In function 'log_module_get_name':
repos/apache-mynewt-core/sys/log/full/src/log.c:236:12: error: implicit declaration of function 'logcfg_log_module_name' [-Werror=implicit-function-declaration]
  236 |     name = logcfg_log_module_name(module);
      |            ^~~~~~~~~~~~~~~~~~~~~~
repos/apache-mynewt-core/sys/log/full/src/log.c:236:10: error: assignment to 'const char *' from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
  236 |     name = logcfg_log_module_name(module);
      |          ^
cc1: all warnings being treated as errors

```
GCC version. 
> /gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-gcc

OS
Linux mint 64-bit

sjanc, 2024-09-27:
Hi,

This seems to work for me on master branch. Could you paste output of 'newt info' command (executed in project repo)

ujur007, 2024-09-27:
yes, here it is.

> Apache Newt 1.12.0 / unknown / 2024-06-18_23:28
> Repository info:
>     * apache-mynewt-core: fd2519971422867204b94786be6b274804495a57, (dirty: local changes)
>     * apache-mynewt-mcumgr: 1c9b642fa3bfb3048b7ed7e75a278667230816c2, 0.4.0
>     * apache-mynewt-nimble: 675452b62822efd3e3fd28ac55eb166baf8461f4, 1.7.0
>     * mcuboot: 304fd41980ed929533b9f387dde1b463b0be5b90, 2.0.0

sjanc, 2024-09-30:
Hi,

Looks like you are using apache-mynewt-core from master branch,  this will not work.

If you want to play with mynewt from master branch you need to used newt tool also from master branch.
NimBLE should likely be also from master branch

ujur007, 2024-09-30:
Yes, I am using a master branch.

sjanc, 2024-11-29:
please reopen if this is stil an issue for you

closed 2024-11-29 by sjanc