apache/mynewt-core #3366: Remove Google Analytics from mynewt Website

niallkp, 2025-01-31:
Hi mynewt Team!

The mynewt website is using _**Google Analytics**_ on **2,239** pages in versions **0.9**, **1.0**, **1.1**, **1.2** & **1.3** of its documentation.

The ASF **_Privacy Policy_**[1][2] does not permit the use of **_Google Analytics_** on any ASF websites and the ASF Infra team will soon enforce a **_Content Security Policy_**(CSP) that will block access to external trackers:

- https://lists.apache.org/thread/w34sd92v4rz3j28hyddmt5tbprbdq6lc

Please could you remove the use of the Google Analytics from the mynewt website?

- https://lists.apache.org/thread/w2jdpmkwd1r0dplltrlvr68ho0q1qkf5

The ASF hosts its own **_Matomo_** instance to provide projects with analytics and you can request a tracking id for your project by sending a mail to **privacy AT apache.org**.

- https://privacy.apache.org/faq/committers.html#can-i-use-web-analytics-matomo

Additionally I would recommend reviewing any external resources loaded by your website. For example:

- fonts.googleapis.com
- maxcdn.bootstrapcdn.com

The Content Security Policy will prevent any resources being loaded from 3rd Party providers that the ASF does not have a Data Processing Agreement (DPA) with. On the 1st February Infra will begin a temporary "brownout" when the CSP will be turned on for a short period. This will allow projects to check which parts, if any, of their websites will stop working. The Privacy FAQ answers a number of questions about which external providers are permitted or not:

- https://privacy.apache.org/faq/committers.html

Thanks

Niall

[1] https://privacy.apache.org/policies/website-policy.html
[2] https://privacy.apache.org/faq/committers.html#can-i-use-google-analytics

niallkp, 2025-01-31:
I created a PR request in the **mynewt-site** repo - see https://github.com/apache/mynewt-site/pull/575

niallkp, 2025-03-11:
Thanks for fixing this - much appreciated

closed 2025-03-11 by niallkp

piotrnarajowski, 2025-04-14:
Hello @niallkp 
Could you please advise on how to properly embed Youtube videos to comply with ASF  Security Policy? As recommended in [FAQ](https://privacy.apache.org/examples/youtube-html/with-youtube-embeds.html), in [PR#578](https://github.com/apache/mynewt-site/pull/578) and [#582](https://github.com/apache/mynewt-site/pull/582)  we've implemented YT video placeholders and .htaccess file that allows reading resources from YouTube, but to no avail. 
Please see https://mynewt.apache.org/talks/ section

piotrnarajowski, 2025-04-18:
Thanks! I've done a few updates and currently .htaccess file is being copied into root of asf-site branch. Unfortunately this doesn't fix the issue on PC, but works ok when I'm visiting the website on my smartphone. Any ideas? Maybe it should be copied into different directory?

niallkp, 2025-04-18:
AIUI, CSP header changes don't cause the cache to be refreshed. I tried the following page which didn't work

- https://mynewt.apache.org/talks/

However, when I try the following link, it does work (I believe this causes a new copy):
- https://mynewt.apache.org/talks/index.html

You can try asking Infra to clear the cache for mynewt or make a minor change to the page(s) affected