apache/mynewt-core #3385: USB mass storage fails to write on Fedora41

jfclere, 2025-03-12:
Based on https://www.codecoup.pl/how-to-build-and-use-drag-and-drop-bootloader-on-mynewt/ I have tried to install an app to a pca10059 (dongle), the mass storage is readable/writeable but it seems the app is not install nor persisted in the dongle.

jfclere, 2025-03-12:
[image]

jfclere, 2025-03-12:
syscfg.vals:
    #Use standard USB descriptor generated by Mynewt
    USBD_STD_DESCRIPTORS: 1
    #Set usb descriptor values
    USBD_VID: 0xC0CA
    USBD_PID: 0xC01B
    USBD_VENDOR_STRING: '"Codecoup"'
    USBD_PRODUCT_STRING: '"Mynewt dongle"'

    #GPIO pin to check during boot for DFU activation.
    MSC_FAT_BOOT_PIN: BUTTON_1
    #GPIO pin value required for DFU activation.
    MSC_FAT_BOOT_PIN_VALUE: 0
    #Set to 1 if boot pin needs internal pull-up resistor.
    #Set to 2 if boot pin needs internal pull-down resistor.
    MSC_FAT_BOOT_PIN_PULL: 1
    #FAT volume name.
    MSC_FAT_VIEW_VOLUME_NAME: '"MYNEWT_BOOT"'
    #Add 'MYNEWT.HTM' file that will redirect to mynewt web page.
    MSC_FAT_VIEW_MYNEWT_SHORTCUT: 1
    #Add git hashes into README.TXT.
    MSC_FAT_VIEW_DEFAULT_README_INCLUDE_HASHES: 1
    #Simulated disk size in kB (Only FAT16 is supported!).
    MSC_FAT_VIEW_DISK_SIZE: 4000
    #Simulated disk sectors per cluster (Only FAT16 is supported!).
    MSC_FAT_VIEW_SECTORS_PER_CLUSTER: 1
    #Add 'Drop image here' file.
    MSC_FAT_VIEW_DROP_IMAGE_HERE: 1
    #Show image from slot 0 as a file 'FIRMWARE.IMG'.
    MSC_FAT_VIEW_SLOT0_IMAGE: 1
    #Enable DFU.
    BOOT_PREBOOT: 1
    MSC_FAT_VIEW_SLOT0_IMAGE: 1
    CONSOLE_UART: 0
    CONSOLE_RTT: 1
    #MSC_FAT_VIEW_LOG_MOD: 1
    MSC_FAT_VIEW_LOG_LVL: 0
    IMG_MGMT_LOG_LVL: 0
    DFLT_LOG_LVL: 0
    # LOG_LEVEL: DEBUG

the targets/nordic_pca10059_boot/syscfg.yml

jfclere, 2025-03-12:
targets/nordic_pca10059_boot/pkg.yml:
```
pkg.name: targets/nordic_pca10059_boot
pkg.type: target
pkg.description:
pkg.author:
pkg.homepage:

pkg.deps:
    - "@apache-mynewt-core/hw/usb/tinyusb/msc_fat_view"
```
targets/nordic_pca10059_boot/target.yml:
```
target.app: "@mcuboot/boot/mynewt"
target.bsp: "@apache-mynewt-core/hw/bsp/nordic_pca10059"
target.build_profile: optimized

```

kasjer, 2025-03-13:
@jfclere, could you please enable showing hidden files. It may be that Fedora creates some additional root entries that eat up root folder entry space. Currently only 16 entries are allowed (just one sector) but when there are names with names that are not XXXXXXXX.EXT (upper case or file name is longer), FAT entry occupies more space.

kasjer, 2025-03-13:
@jfclere You need to add line
```yml
    MSC_FAT_VIEW_UPDATE_HANDLER: 1
```
This was enabled by default previously now it's not.
Let me know if this fixes your problem.

jfclere, 2025-03-13:
Yes that helps
`00000 msc_fat_view_write_unallocated_sector: 0x2000070c`

The img is now on the dongle.

> [@jfclere](https://github.com/jfclere) You need to add line
> 
>     MSC_FAT_VIEW_UPDATE_HANDLER: 1
> 
> This was enabled by default previously now it's not. Let me know if this fixes your problem.

jfclere, 2025-03-13:
There is still something wrong, I type` sync` or try to `umount` , and my box complains:
```
[845234.757966] usb 3-5.1.1: new full-speed USB device number 46 using xhci_hcd
[845234.858734] usb 3-5.1.1: New USB device found, idVendor=c0ca, idProduct=c01b, bcdDevice= 1.00
[845234.858737] usb 3-5.1.1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[845234.858738] usb 3-5.1.1: Product: Mynewt dongle
[845234.858739] usb 3-5.1.1: Manufacturer: Codecoup
[845234.858740] usb 3-5.1.1: SerialNumber: 1
[845234.865106] usb-storage 3-5.1.1:1.0: USB Mass Storage device detected
[845234.865277] scsi host0: usb-storage 3-5.1.1:1.0
[845235.905205] scsi 0:0:0:0: Direct-Access     Mynewt   Mass Storage     1.0  PQ: 0 ANSI: 2
[845235.905508] sd 0:0:0:0: Attached scsi generic sg0 type 0
[845235.907189] sd 0:0:0:0: [sda] Media removed, stopped polling
[845235.907633] sd 0:0:0:0: [sda] Attached SCSI removable disk
[845235.915268] sd 0:0:0:0: [sda] 8000 512-byte logical blocks: (4.10 MB/3.91 MiB)
[845235.916504] sda: detected capacity change from 0 to 8000
[845235.937754]  sda:
[845296.116536] usb 3-5.1.1: reset full-speed USB device number 46 using xhci_hcd
[845311.447242] usb 3-5.1.1: device descriptor read/64, error -110
[845327.319222] usb 3-5.1.1: device descriptor read/64, error -110
[845327.509105] usb 3-5.1.1: reset full-speed USB device number 46 using xhci_hcd
[845343.201895] usb 3-5.1.1: device descriptor read/64, error -110
[845359.073922] usb 3-5.1.1: device descriptor read/64, error -110
[845359.271928] usb 3-5.1.1: reset full-speed USB device number 46 using xhci_hcd
[845374.849657] usb 3-5.1.1: device descriptor read/8, error -110
[845390.209548] usb 3-5.1.1: device descriptor read/8, error -110
[845390.396699] usb 3-5.1.1: reset full-speed USB device number 46 using xhci_hcd
[845406.080502] usb 3-5.1.1: device descriptor read/8, error -110
[845421.440397] usb 3-5.1.1: device descriptor read/8, error -110
[845421.543159] usb 3-5.1.1: USB disconnect, device number 46
[845421.550179] device offline error, dev sda, sector 33 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2
[845421.550183] Buffer I/O error on dev sda, logical block 33, lost async page write
[845421.695094] usb 3-5.1.1: new full-speed USB device number 47 using xhci_hcd
[845437.398072] usb 3-5.1.1: device descriptor read/64, error -110
[845453.270903] usb 3-5.1.1: device descriptor read/64, error -110
[845453.459856] usb 3-5.1.1: new full-speed USB device number 48 using xhci_hcd
[845469.152806] usb 3-5.1.1: device descriptor read/64, error -110
[845485.024729] usb 3-5.1.1: device descriptor read/64, error -110
[845485.125958] usb 3-5.1-port1: attempt power cycle
[845485.760550] usb 3-5.1.1: new full-speed USB device number 49 using xhci_hcd
[845501.311592] usb 3-5.1.1: device descriptor read/8, error -110
[845516.671431] usb 3-5.1.1: device descriptor read/8, error -110
[845516.860209] usb 3-5.1.1: new full-speed USB device number 50 using xhci_hcd
[845532.543080] usb 3-5.1.1: device descriptor read/8, error -110
[845547.903086] usb 3-5.1.1: device descriptor read/8, error -110
[845548.006501] usb 3-5.1-port1: unable to enumerate USB device
[845548.053674] nautilus: attempt to access beyond end of device
                sda: rw=0, sector=0, nr_sectors = 1 limit=0
[845548.053682] FAT-fs (sda): unable to read boot sector to mark fs as dirty
```
And the dongle needs a reset.

jfclere, 2025-03-13:
it seems we are calling:
 ```
          if (BOOT_LOADER) {
                img_mgmt_state_confirm();
                hal_system_reset();
```
Without "cleaning" the FAT we created, is that correct?

kasjer, 2025-03-13:
System just written file to this FAT formatted driver and it wants to flip a bit on boot sector to mark disk as clean so next time it is inserted in Windows there will be no warning that there is a problem.
We reboot to finish the upgrade. Linux will just observe this as driver reconnect and dirty bit in the boot sector is reported as clean.
During write of the files, Linux will write boot sector (with just one bit changed) several time fat_view just ignores write to this sector.

jfclere, 2025-03-14:
OK thanks

closed 2025-03-14 by jfclere